pa3gsb/Radioberry-2.x #5: FPGA options

dwhacks, 2020-04-11:
I know it says in the wiki 10CL016YE144C8G and 10CL025 are possible, but what about parts ending in I7G?
The standard part doesn't seem to be in stock anywhere, and the I7G part looks compatible as far as I can tell. and still less expensive then the 25 part.

10CL016YE144I7G
https://www.mouser.ca/ProductDetail/Intel-Altera/10CL016YE144I7G?qs=sGAEpiMZZMvoScKlWpK8TEXJPA3QSkS0VjYIq3om7tY=

Thanks, really looking forward to building this as i've had the PCB for some time now.

pa3gsb, 2020-04-11:
I think it is possible to use. I think a variant of the gateware must be made.

dwhacks, 2020-04-11:
Well It seems the price has gone up anyway and the 25 part is very close to the same cost. It could be nice if we knew exactly which were compatible. FPGA's are very new for me so I'm learning. I will probably order the 10CL025YE144C8G if we are sure that is compatible.

Thanks a lot, really hoping to get this going soon!

closed 2021-09-26 by pa3gsb