pa3gsb/Radioberry-2.x #8: Question re the beta 4 board

ravimiranda, 2020-12-07:
Hi, 
I wanted to experiment with the beta4 version of the board, so sent it to JLCPCB who have pointed out that there is an error with the PCB at two points (at least). 

JLCPCB say the following "Since we can not make blind and buried hole, we only can fabricate the board follow one drilling layer , please kindly check and advise"

There are three .drl files Radioberry.drl, Radioberry-NPTH.drl and Radioberry-PTH.drl. Are all the three files required for fabrication.

Has anyone else faced this issue of fabrication and what steps were taken to resolve this please?
[image]

I've attached the image with the vias(?) marked by JLCPB if it helps.

Please let me know how to solve this as I'm not an expert in PCB design.

/RM

pa3gsb, 2020-12-07:
RM

There are already PCB created based on the release files...

I have generated the drill files again (now available in the repo in the beta4 release)
'
The radioberry.drl file must do the job.

Please let me know if JLCPCB now accepts the files?

73 Johan
PA3GSB

closed 2020-12-07 by pa3gsb

ravimiranda, 2020-12-08:
Thanks Johan,
I have now submitted the new set of gerbers to JLCPCB again, and they seem to have been accepted. 
Thank you for your help.

Kind regards,
/Ravi

pa3gsb, 2020-12-09:
Ravi

Tnx for the feedback!

Have in building this nice little radio.

73 Johan
PA3GSB