pa3gsb/Radioberry-2.x #12: Raspbian OS Kernel for stable release.

omarlira77, 2021-08-11:
Hi Johan, hoping going well.

I am looking to run the stable release, without sucessfull results.

I have tried with:

kernel 5.10.17
kernel 5.4.79
kernel 4.19.197
kernel 4.19.18

all are **Raspbian 10 Buster**.

When you remark: 
> "The latest linux distro is not working with the stable version... so in case of setup problems you need to use the develop version."

Do you refer to any specific kernel version?, or Raspbian version? (buster, stretch, jessie).

Can you please help us, specifying the **version of raspian or kernel, to run the stable version**?

Thanks.

73 Omar
XE1SWL Omar

pa3gsb, 2021-09-26:
Omar,

I did miss this issue....

Is the question still open ?

Please use the RASPBERRY PI OS

update and upgrade and then install the release.

73 Johan

madpsy, 2021-09-27:
The important there is to do...

sudo apt update
sudo apt upgrade
reboot

...then run the installer. The script installs the latest kernel headers, which won't match the running kernel if you haven't done a full upgrade first.

closed 2021-09-27 by pa3gsb