pa3gsb/Radioberry-2.x #14: piHPSDR dramatic breakage

madpsy, 2021-09-27:
Just complied the latest version of piHPSDR and it's working perfectly with an HL2.

However, with the radioberry it goes crazy. See attached screenshots showing a pretty rainbow and excessive CPU usage of both pihpsdr and the radioberry user service.

I'm unsure if this is an issue with piHPSDR, how I compiled it or a bug being hit in the radioberry driver. Once this has happened, no other software can use the radioberry (such as SparkSDR) until you run:

1) service radioberry stop
2) rmmod radioberry ; modprobe radioberry
3) service radioberry start

Having to restart the kernel module suggests something is going horribly wrong, somewhere, I'm just not sure where!

I am using the latest stable release of the radioberry driver.

[image]
[image]

madpsy, 2021-09-27:
Okay so the issue is PureSignal.

If piHPSDR is compiled with PS support but even if it is **not enabled in piHPSDR** this occurs. I see you disable it in your install script

`sed -i '/^PURESIGNAL_INCLUDE=PURESIGNAL/c\#PURESIGNAL_INCLUDE=PURESIGNAL' ./Makefile;`

I am curious why this breaks radioberry is such a dramatic fassion.

pa3gsb, 2021-09-27:
please do not use the gpio in pihpsdr

which version of radioberry are you running?

madpsy, 2021-09-27:
Ah ok - GPIO is the issue, not PureSignal. My bad. I use PS with the HL2 so glad it can be remain there.

I have recompiled with MIDI and PS enabled but GPIO disabled and appears to be working.

I am currently using the beta build of the radioberry driver as I use your pre-amp.

Thanks

closed 2021-09-27 by pa3gsb