pa3gsb/Radioberry-2.x #28: Bandwidth

paulh002, 2022-05-28:
Hi All,

The current SoapyRadioberry driver returns a number of bandwidth which are the same as the sampling frequencies.
If I compare this with other SoapySDR drivers than this is not correct. For example Sdrplay uses bandwidth to indicate the filter options that are  placed after the frontend mixer. Because radioberry does not use a frontend mixer, my suggesten is to remove all bandwidths from de Soapyradioberry driver. If this is fine, I will issue a pull request.

pa3gsb, 2022-05-29:
Hi Paul,

The radioberry is based on soapysdr.  

You are talking to remvove:
		double getBandwidth( const int direction, const size_t channel ) const;
		std::vector<double> listBandwidths( const int direction, const size_t channel ) const;

The only remark i have to make : do we not break the idea of a generic soapy interface wich expects these methods?

73 Johan

paulh002, 2022-05-29:
> std::vector listBandwidths( const int direction, const size_t channel )

If you don't overload the functions SoapySDR will just return empty vectors.
Or we can keep the functions and return empty vector, this is how I did it
now.

Right now I can only test it with my own software, normally you choose a
bandwidth which is lower than the half the sample rate.
It is not such an issue, I will try to test it with some other generic
software to see if it works.

73,
Paul

pa3gsb, 2022-05-29:
Paul,

Returning empty list make sense... user of the api can remove or disable the option of selecting the bandwith.

73 Johan

closed 2023-10-14 by pa3gsb