pa3gsb/Radioberry-2.x #36: Installing Radioberry driver... modprobe: FATAL: Module radioberry not found

seelenreiter, 2023-07-28:
Hi,

i bought a Radio Berry V5.0 Board from Aliexpress. "Radio berry v5.0 software definierte Funkgeräte ad9866 10 cl016 für himbeer pi4b rpi4"

I installed a fresh Raspian OS installation (i tried both 32 and 64)
I did the apt updat & upgrade and a firmware update.

now, i try to install radioberry_install.sh:

**============================================
Radioberry software installation.

You will install the following versions: 

	Gateware version 73.2
	Driver version 0.92
	Firmware version 2021.09.10
============================================

--- Installing dependency: raspberrypi-kernel-headers
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1:1.20230405-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
--- Installing dependency: linux-headers-rpi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-headers-rpi
--- Installing dependency: git
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.30.2-1+deb11u2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
--- Installing dependency: device-tree-compiler
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
device-tree-compiler is already the newest version (1.6.0-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
--- Installing dependency: pigpio
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
pigpio is already the newest version (1.79-1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Cloning into 'Radioberry-2.x'...
remote: Enumerating objects: 1021, done.
remote: Counting objects: 100% (1021/1021), done.
remote: Compressing objects: 100% (700/700), done.
remote: Total 1021 (delta 407), reused 759 (delta 279), pack-reused 0
Receiving objects: 100% (1021/1021), 28.47 MiB | 16.79 MiB/s, done.
Resolving deltas: 100% (407/407), done.
Removed /etc/systemd/system/multi-user.target.wants/radioberry.service.
Installing Radioberry gateware Cyclone 10 CL016...

Radioberry gateware Cyclone 10 CL016 installed.
Installing Radioberry driver...
modprobe: FATAL: Module radioberry not found.
make[1]: *** /lib/modules/6.1.41-v8+/build: No such file or directory.  Stop.
make: *** [Makefile:6: radioberry.ko] Error 2
cp: cannot stat 'radioberry.ko': No such file or directory
modprobe: FATAL: Module radioberry not found in directory /lib/modules/6.1.41-v8+
chmod: cannot access '/dev/radioberry': No such file or directory
modinfo: ERROR: Module radioberry not found.

Radioberry driver installed.
Installing Radioberry firmware...
[... 142 lines cut ...]
radioberry.c:542:6: warning: conflicting types for ‘start_timer_thread’
  542 | void start_timer_thread() {
      |      ^~~~~~~~~~~~~~~~~~
radioberry.c:200:2: note: previous implicit declaration of ‘start_timer_thread’ was here
  200 |  start_timer_thread();
      |  ^~~~~~~~~~~~~~~~~~
gcc -o radioberry register.o bias.o measure.o radioberry.o  -lrt -lm -lpthread 

Radioberry firmware installed.
Installing radioberry service ...
Radioberry service installed.
Created symlink /etc/systemd/system/multi-user.target.wants/radioberry.service → /etc/systemd/system/radioberry.service.

============================================
Radioberry software is installed!
Have fun using the SDR Radioberry

73 Johan PA3GSB
============================================**

sudo modinfo radioberry

modinfo: ERROR: Module radioberry not found.

Any hint?

LoyalServant, 2023-08-01:
The simple answer is that the kernel headers are missing.
I am trying to figure out how to resolve it since the package manager is brain-dead.

if you run 'uname -a' it will tell you the kernel version.

in my case.... 

`
mark@raspberrypi:/lib/modules $ uname -a
Linux raspberrypi 6.1.42-v8+ #1668 SMP PREEMPT Thu Jul 27 15:34:18 BST 2023 aarch64 GNU/Linux
`

So, its expecting a symlink to the kernel headers for this version.
This is why you get a ...
`
make[1]: *** /lib/modules/6.1.41-v8+/build: No such file or directory. Stop.
`

That's because you need a symlink from [version]/build -> /usr/src/linux-headers-[version]
But, it's not installed. This seems to be a chronic problem in my googling that the raspberry pi doesn't seem to keep the kernel sources in sync with what the system is running..... things get disconnected.

So........
`
root@raspberrypi:/# apt-get install raspberrypi-kernel-headers
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1:1.20230405-1).
`

Yep. the package manager has lost the plot. (yes, I already tried numerous updates)
I am going to try to wipe the card and reimage.....

LoyalServant, 2023-08-01:
@n7gtb I bet that's exactly what the issue is... but I reimaged already and didn't get to try that. Thanks.

I did get it to work when I reimaged....
So, I reimaged the card using the image '2023-05-03-raspios-bullseye-arm64' downloaded from the website with the raspberry pi imager... created my user, set timezone, etc....
Then apt-get update.
Then apt-get install raspberrypi-kernel-headers
Verifed that the symlink from 'build' was in fact created, so the correct headers were now installed and matched uname -a.
Module build proceeeded as expected...

Now, the service configuration expects a 'pi' user and group. I didn't create this as my user at setup.
I simply changed the user/group in radioberry.service, reloaded systemd, and then I was able to start the service.

This shoud probably be mentioned in the wiki that the scripts expect a 'pi' user and need to be changed depending on whatever user you create. As I am not a regular pi user I am out of my element... and wonder if this is some change where this 'pi' user is no longer created....

seelenreiter, 2023-08-02:
Thank you, i'll do a retry.
Well, i used '2023-05-03-raspios-bullseye-arm64' and User pi...

seelenreiter, 2023-08-02:
Thank you, it's working now.

It seems to be a temporary problem in 2023-05-03-raspios-bullseye-arm64.
now it is fixed.

closed 2023-08-02 by seelenreiter