pa3gsb/Radioberry-2.x #37: Extra (physical) controls - I/O interface

patrickomatik, 2023-08-03:
I want to add custom controls - at least a rotary encoder (for tuning), and possibly also a 7 segment display (for frequency display) and maybe an rtc/gps. Is there any free I/O on the pi4 with the pi-board attached? I see i2c is used by the piberry, presumably I can piggyback on that? And the SPI port is, I think, being used as generic I/O so is out of bounds? I have seen the pinout on the wiki, but I don't want to do anything stupid that could damage the (valuable, rare!) piberry - so guidance would be most welcome.

LoyalServant, 2023-08-09:
I haven't studied the code that much..... maybe an hour or so. Someone may have a better answer.
The SPI interface is used by the radioberry. I2C from the raspberry pi passes straight thru.
I think your best bet based on what I have seen so far is a microcontroller on the I2C bus.
Do all of your user interactions and additional display stuff there, and query the microcontroller at an interval to see if any button was pressed, control rotated, etc.
There are 4 pins on the FPGA unused attached to CN301, but you would have to implement those yourself in the gateware.

pa3gsb, 2023-10-14:
use of rpi io ports see: https://github.com/pa3gsb/Radioberry-2.x/wiki/GPIO-RPI---Radioberry

closed 2023-10-14 by pa3gsb