pa3gsb/Radioberry-2.x #39: Radioberry installation failure

w9ran, 2023-08-31:
I'm trying to install Radioberry on a brand-new Pi 4.0 Bullseye system that was created using the Pi Imager, with only a pi user account.   Radioberry hardware is V 2.0 Beta 4.   Followed the installation process here:   https://github.com/pa3gsb/Radioberry-2.x/releases

Testing the installation with sudo modinfo radioberry fails with modinfo: ERROR: Module radioberry not found.  

Examination of the install script terminal shows one problem when installing dependencies: 

--- Installing dependency: linux-headers-rpi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-headers-rpi

The driver installation fails as shown in the terminal snippet below:  (I can provide the complete terminal contents if anyone would like to look at it).

---

Installing Radioberry gateware Cyclone 10 CL025...
Radioberry gateware Cyclone 10 CL025 installed.

Installing Radioberry driver...
modprobe: FATAL: Module radioberry not found.
make[1]: *** /lib/modules/6.1.21-v8+/build: No such file or directory.  Stop.
make: *** [Makefile:6: radioberry.ko] Error 2
cp: cannot stat 'radioberry.ko': No such file or directory
modprobe: FATAL: Module radioberry not found in directory /lib/modules/6.1.21-v8+
chmod: cannot access '/dev/radioberry': No such file or directory
modinfo: ERROR: Module radioberry not found.
Radioberry driver installed.
Installing Radioberry firmware...
(lots of gcc warnings omitted)
Radioberry firmware installed.
Installing radioberry service ...

Radioberry service installed.

---

My understanding is that the radioberry installation script does everything but the needed kernel module is missing. and I'm not sure what to do about it.    Any suggestions would be appreciated.

I have installed pihpsdr and proved that it works using a different H-L on the network, but with it disconnected, discovery naturally comes up empty.

Thanks and 73,

Bob W9RAN

SpudGunMan, 2023-08-31:
I found that .. in a "new install" you need to try adding "arm_64bit=0" to config.txt, reboot and try again 

specifically your kernel header directories like build/ are missing and the automation script for driver install on this project will hang and prevent build of .ko

RichNeese, 2023-08-31:
Iss there a plan to fix this for arm64. as its default now. we should noth ave to disable arm64

SpudGunMan, 2023-08-31:
In my very little research this is a workaround for a rpi-foundation OS decisions. 

The install script here is performing automation tasks, while a check could be put in easily... 

The larger issue is, where is the proper bits and folders not part of the kernel-headers deb in the first place.. 

(I think from my limited digging)

w9ran, 2023-08-31:
That definitely helped - the installation completed and I now get a proper response from modprobe.  I can start and stop radioberry using the command line and it shows the gateware is loaded (ver 73.3) - but unfortunately the radioberry isn't working.  All I get is this screen, with no audio or signals present and "sequence error" displayed:
[image]

Any help would be appreciated!
Bob

w9ran, 2023-08-31:
Arggh!  You're right:
E: Unable to locate package linux-headers-rpi
So on to that issue...
Bob

SpudGunMan, 2023-08-31:
Can you connect with spark or thetis ?

w9ran, 2023-08-31:
Sheesh, my troubleshooting brain is half-asleep today, I hadn't thought of that.   But YES I can!   Thetis was chugging somewhat but Spark runs nicely with it (probably just the different sample rates).    As RichNeese pointed out, there's still a missing package so hopefully once that is corrected it will run locally as well.   Many thanks!
Bob

w9ran, 2023-08-31:
The installation terminal still has this error:
E: Unable to locate package linux-headers-rpi

I tried to update this package but it says the latest is already installed:  sudo apt install raspberrypi-kernel-headers - evidently that's a different package.    Evidently linux-headers-rpi is needed for compiling a kernel module.

Both Radioberry and pihpsdr on this Pi4 work separately - but not at the same time.   I can control the Radioberry from another application such as Spark, and I can control a different Hermes-Lite from pihpsdr on the Radioberry Pi  - but as soon as I try to do both at the same time, the stream locks up and the Radioberry shows a solid waterfall spike.   I have to reboot the Pi in order to clear this condition.    

The  linux-headers-rpi error is the only one I can find - any suggestions about how to fix that would be appreciated, thanks.
Bob

closed 2024-02-05 by pa3gsb