pa3gsb/Radioberry-2.x #49: Issue getting software and driver installed on Debian BookWorm

stevegg, 2024-01-08:
sudTrying to install the Radioberry software and driver on bookworm and running into issues.  Everything installs ok but when I restart the device (/dev/radioberry) disappears.  If I then re-run the install script in releases it will reappear until the next reboot.

`uname -a
Linux openwebrx.local 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

pi@openwebrx:~ $ sudo radioberry

====================================================================
====================================================================
			Radioberry V2.0

	Supports 4 receivers and 1 transmitter.

	Build version: 2022.01.24

	Have fun Johan PA3GSB

	Report requests or bugs to <pa3gsb@gmail.com>.
====================================================================
====================================================================
Failed to open /dev/radioberry: No such file or directory`

Here's what I get from running the install script:

`============================================
Radioberry software installation.

You will install the following versions:

	Gateware version 73.3
	Driver version 0.94
	Firmware version 2022.01.24
============================================

Install CL016 or CL025 radioberry version: 1 = CL016 or 2 = CL025? 2
--- Installing dependency: raspberrypi-kernel-headers
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1:1.20230405-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
--- Installing dependency: linux-headers-rpi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-headers-rpi
--- Installing dependency: git
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.39.2-1.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
--- Installing dependency: device-tree-compiler
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
device-tree-compiler is already the newest version (1.6.1-4+b1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
--- Installing dependency: pigpio
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
pigpio is already the newest version (1.79-1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Cloning into 'Radioberry-2.x'...
remote: Enumerating objects: 1089, done.
remote: Counting objects: 100% (1089/1089), done.
remote: Compressing objects: 100% (752/752), done.
remote: Total 1089 (delta 429), reused 819 (delta 293), pack-reused 0
Receiving objects: 100% (1089/1089), 30.46 MiB | 8.75 MiB/s, done.
[... 161 lines cut ...]
Radioberry, Starting packet control part.
Radioberry, Starting packet tx part.`

Everything works fine....but as soon as I reboot the /dev/radioberry will disappear and nothing will work.

`pi@openwebrx:~ $ sudo radioberry

====================================================================
====================================================================
			Radioberry V2.0

	Supports 4 receivers and 1 transmitter.

	Build version: 2022.01.24

	Have fun Johan PA3GSB

	Report requests or bugs to <pa3gsb@gmail.com>.
====================================================================
====================================================================
Failed to open /dev/radioberry: No such file or directory`

What am I missing?

Steve

stevegg, 2024-01-08:
Was an issue loading the module.

closed 2024-01-08 by stevegg

RattyDAVE, 2024-01-18:
I am having this error too. If you look at the output above the main problem seems to be the line

`E: Unable to locate package linux-headers-rpi`