pa3gsb/Radioberry-2.x #53: make fails on RPI4

oneofthejonesboys, 2024-03-01:
Just bought a new Radioberry-2.x. Freshly installed and update OS. Software failed during make command. Error message:

owner@radioberry:/tmp/Radioberry-2.x $ make -j$(nproc)
bash -c 'cd SBC/rpi-4/device_driver/driver && make'
make -C wdsp
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
cc -I /include -I /include/linux  -g -fPIC -O3 -D _GNU_SOURCE	   -c -o linux_port.o linux_port.c
cc -I /include -I /include/linux  -g -fPIC -O3 -D _GNU_SOURCE	   -c -o amd.o amd.c
cc -I /include -I /include/linux  -g -fPIC -O3 -D _GNU_SOURCE	   -c -o ammod.o ammod.c
make[2]: *** /lib/modules/6.1.0-rpi8-rpi-v8/build: No such file or directory.  Stop.
make[1]: *** [Makefile:6: radioberry.ko] Error 2
make: *** [Makefile:26: SBC/rpi-4/device_driver/driver/radioberry.ko] Error 2
make: *** Waiting for unfinished jobs....
cc -I /include -I /include/linux  -g -fPIC -O3 -D _GNU_SOURCE	   -c -o amsq.o amsq.c
In file included from linux_port.c:29:
comm.h:50:10: fatal error: fftw3.h: No such file or directory
   50 | #include "fftw3.h"
      |          ^~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:238: linux_port.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from amd.c:27:
comm.h:50:10: fatal error: fftw3.h: No such file or directory
   50 | #include "fftw3.h"
      |          ^~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:238: amd.o] Error 1
In file included from amsq.c:27:
comm.h:50:10: fatal error: fftw3.h: No such file or directory
   50 | #include "fftw3.h"
      |          ^~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:238: amsq.o] Error 1
In file included from ammod.c:27:
comm.h:50:10: fatal error: fftw3.h: No such file or directory
   50 | #include "fftw3.h"
      |          ^~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:238: ammod.o] Error 1
make: *** [Makefile:53: wdsp/libwdsp.so] Error 2
owner@radioberry:/tmp/Radioberry-2.x $ ^C

OS info:
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

I looked in the /lib/modules/6.1.0-rpi8-rpi-v8/ directory and indeed...there is no build dir. I added it but this did not solve all the problems.

Any suggestions to resolve this would be greatly appreciated. 

Jim
KF0BXC

RattyDAVE, 2024-03-02:
Just to add this have the same problem on a Raspberry Pi 5. With both the 32 and 64 bit versions.

RattyDAVE, 2024-03-05:
I have tried

```
Raspberry Pi OS with desktop
Release date: December 5th 2023
System: 32-bit
Kernel version: 6.1
Debian version: 12 (bookworm)
```

and 

```
Raspberry Pi OS with desktop
Release date: December 5th 2023
System: 64-bit
Kernel version: 6.1
Debian version: 12 (bookworm)
```

On both the RPI4 and RPI5 and unable to compile.

```
/lib/modules/6.1.0-rpi7-rpi-v8/build
/lib/modules/6.1.0-rpi8-rpi-v8/build
```

Are missing.

Any idea on how we can get a working firmware?

bpbeatty, 2024-03-07:
On the RPI4 I was not able to build the latest from master. However by setting `arm_64bit=0` in the [config.txt](https://www.raspberrypi.com/documentation/computers/config_txt.html) I was able to use the [32-bit kernel modules](https://www.raspberrypi.com/documentation/computers/config_txt.html#arm_64bit) to build the software using [this release](https://github.com/pa3gsb/Radioberry-2.x/releases/tag/radioberry).
Unfortunately I believe this will not work on the RPP5 as it only supports a 64-bit kernel.

qrp73, 2024-03-12:
If you're used rpi-update to update kernel version, it don't have kernel headers, so you will be unable to compile driver for such kernel.

You can compile it for both 64-bit and for 32-bit kernel by restoring current stock kernel with:
```
sudo apt install --reinstall raspi-firmware
```

Then reboot the system and install kernel headers:

For 64-bit rpi-4:
```
sudo apt install linux-headers-rpi-v8
```

For 32-bit rpi-4:
```
sudo apt install linux-headers-rpi-v7l
```

But today apt update (12-MAR-2024) rolls out a new kernel 6.6 which definitions in kernel headers are not compatible with current radioberry driver code. So currently there is no way to compile driver neither for 32-bit nor for 64-bit version. The driver code needs to be fixed for latest kernel changes.

pa3gsb, 2024-04-23:
solved

closed 2024-04-23 by pa3gsb