pa3gsb/Radioberry-2.x #54: cannot compile driver for the latest kernel 6.6 changes

qrp73, 2024-03-12:
Today raspberry pi rolled out major apt update which upgrades kernel to 6.6.
It breaks radioberry driver and it cannot be compiled anymore for both 32 and 64 versions.

compilation has 3 errors:
**1) too many arguments for class_create():**
```
/usr/src/linux-headers-6.6.20+rpt-common-rpi/include/linux/device/class.h:230:54: note: expected ‘const char *’ but argument is of type ‘struct module *’
  230 | struct class * __must_check class_create(const char *name);
      |                                          ~~~~~~~~~~~~^~~~
/home/pi/SRC/radioberry/Radioberry-2.x/SBC/rpi-4/device_driver/driver/radioberry.c:280:26: error: too many arguments to function ‘class_create’
  280 |    radioberryCharClass = class_create(THIS_MODULE, CLASS_NAME);
      |                          ^~~~~~~~~~~~
```

**2 and 3) functions gpio_export() and gpio_unexport() are not available anymore.** It seems that it requires to use gpiod_export and gpiod_unexport but it uses different argument type and is not clear on how to use it:
```
/home/pi/SRC/radioberry/Radioberry-2.x/SBC/rpi-4/device_driver/driver/radioberry.c:304:9: error: implicit declaration of function ‘gpio_export’; did you mean ‘gpiod_export’? [-Werror=implicit-function-declaration]
  304 |         gpio_export(gpioRxSamplesiPin, false);
      |         ^~~~~~~~~~~
      |         gpiod_export
```

```
/home/pi/SRC/radioberry/Radioberry-2.x/SBC/rpi-4/device_driver/driver/radioberry.c:337:5: error: implicit declaration of function ‘gpio_unexport’; did you mean ‘gpiod_unexport’? [-Werror=implicit-function-declaration]
  337 |     gpio_unexport(gpioRxSamplesiPin);
      |     ^~~~~~~~~~~~~
      |     gpiod_unexport
```

Any info on how to fix it?

paulh002, 2024-03-14:
The gpio api used is deprecated, should be changed to newer API. 
I don't know if Johan already did this, I have made an update, but still testing it. 
There are some more kernel changes that need to be checked.

```
// line 83 add:
        static struct gpio_desc *gpio_desc;  // top of the file
```

```
// line 300 replace gpio part with :	
	gpio_desc = gpio_to_desc(gpioRxSamplesiPin);
	if (!gpio_desc) {
        printk("Failed to get GPIO descriptor\n");
        return -ENODEV;
    }
	retval = gpiod_direction_input(gpio_desc);
    if (retval) {
        printk("Failed to set GPIO pin direction\n");
        return retval;
    }
	
	printk(KERN_INFO "Radioberry: The rx sample state is currently: %d\n", gpio_get_value(gpioRxSamplesiPin));

	// GPIO numbers and IRQ numbers are not the same! This function performs the mapping for us
	// Get the IRQ number for the GPIO pin
    irqNumber = gpiod_to_irq(gpio_desc);
    if (irqNumber < 0) {
        printk("Failed to get IRQ number for GPIO pin\n");
        return irqNumber;
    }
```

``
// line 351 replace gpio_unexport with:
gpiod_put(gpio_desc);
``

qrp73, 2024-03-15:
could you please share all changes?

I tried to apply changes mentioned above to my repository: https://github.com/qrp73/Radioberry-2.x
But it don't works, the driver is crashed during system boot.
I'm using v8 kernel (aarch64 6.6)

paulh002, 2024-03-15:
I made a fix for kernel rpi6.6.y it is available in my fork of Radioberry 2.x  https://github.com/paulh002/Radioberry-2.x
I did not update the installation scripts, because I am not using those also I am not using the pihpsdr software, I use my own.
Add the dtoverlay=radioberry is /boot/firmware/config.txt
If there are issues you can report them in github on the fork in issues. If all is ok I will create a pull request for Johan. 

Copy all the files from SBC/rpi-4/device_driver/driver
execute:

cd Radioberry-2.x/SBC/rpi-4/device_driver/driver
rm *.ko
make
sudo cp radioberry.ko /lib/modules/$(uname -r)/kernel/drivers/sdr

sudo dtc -@ -I dts -O dtb -o radioberry.dtbo radioberry.dts
sudo cp radioberry.dtbo /boot/overlays

sudo reboot

qrp73, 2024-03-15:
@paulh002 I tried your changes in radioberry.c and radioberry.dts, with these changes driver is loaded ok and firmware can read gateware version, but it don't send samples to SDR application.

paulh002, 2024-03-15:
I am not using the firmware in my setup. Make sure the firmware has root privilege or allow access to the driver. I think for test you can run the firmware from commandline like sudo radioberry. Please share the output to analyse

qrp73, 2024-03-15:
I'm running the firmware from console, it starting as usual and shows correct gateware version. And I can connect with SDR application (pihpsdr and SDR++). But it don't send samples. 

I'm using my mod for SDR++ with my HPSDR plugin: https://github.com/qrp73/Radioberry-2.x
It allows to see measured downstream sample rate, I added it for debugging purposes.
And it shows 0 kS/s. The firmware don't send samples.

I tried to reboot into old kernel image with working driver and it works with exactly the same conditions.
I checked I2C it is enabled in Raspberry Pi configuration.

I think something is wrong with pins configuration...

paulh002, 2024-03-15:
====================================================================
====================================================================
                        Radioberry V2.0

        Supports 4 receivers and 1 transmitter.

        Build version: 2022.01.24

        Have fun Johan PA3GSB

        Report requests or bugs to ***@***.***>.
====================================================================
====================================================================
Radioberry gateware version 73-3.
No Alex, N2ADR or generic filters interface board connected to radioberry
Radioberry amplifier config failed; only a problem if amplifier is
installed.
Radioberry, Starting packet rx part.
Radioberry, Starting packet control part.
Radioberry, Starting packet tx part.
Discovery packet received
SDR Program IP-address 192.168.88.17
Discovery Port 20000
Start Port 20000
SDR Program sends UDP Start command
Your Radioberry is registered: http://www.pa3gsb.nl/radioberry/api/read.php

works fine for me. I test now with SDR Console

qrp73, 2024-03-15:
Yes, it started ok and shows correct gateware version (which means it can receive it from gateware).
But it don't receive any samples from device. The software shows black waterfall and my stat in software shows that there is no EP6 UDP packets from firmware.

paulh002, 2024-03-15:
I do receive samples. So it has to be something local.

qrp73, 2024-03-15:
it works, I tried to reboot after reinstalling the driver and now it sends samples :)
Thanks for help

paulh002, 2024-03-15:
Yes, it needs a reboot

qrp73, 2024-03-15:
It works, but I think there is some issue with your code. 

I notice that radioberry_init() is called before radioberry_probe when loading driver on the fly with ```sudo modprobe radioberry```.

It means that this call:
```
result = request_irq(irqNumber,             
						(irq_handler_t) radioberry_irq_handler, 
						IRQF_TRIGGER_RISING,   // Interrupt on rising edge  RQF_TRIGGER_RISING
						"radioberry_rx_irq",    // Used in /proc/interrupts to identify the owner
					NULL);
```
will be executed with not initialized irqNumber. Isn't it?

Probably this is the reason why it didn't send samples in my case. What is your thoughts?

qrp73, 2024-03-15:
as I read on internet, radioberry_init is called first, before radioberry_probe. That explains why it didn't worked for me. Is it possible to move request_irq() call to radioberry_probe() function?

paulh002, 2024-03-15:
Yes, you're right. We have to move the initialization to the probe function.
The init is always executed even when probe fails.
I will change it'

paulh002, 2024-03-15:
I moved the code, you can have a look

qrp73, 2024-03-15:
yes, it works.
I also added some minor fixes to radioberry.c, see my repo: https://github.com/qrp73/Radioberry-2.x/blob/master/SBC/rpi-4/device_driver/driver/radioberry.c

qrp73, 2024-03-16:
I added support for both kernel version, so it can be compiled for old 6.1 and for 6.6 kernel: 
https://github.com/qrp73/Radioberry-2.x

paulh002, 2024-03-16:
@qrp73 can you test the latest driver version? We changed to polling instead of interrupts. This solved a stability issue I had using SoapyRadioberry

keyboarderror, 2024-03-17:
@paulh002 Test report. I found this thread when my Radioberry went dead after a reboot and the 6.6 kernel took effect. I've tried your repo and instructions, rebooted, then ran pa3qsb's install script and  it's working again. I mostly use it with SDRConsole. Thank You.

pa3gsb, 2024-04-23:
solved