pa3gsb/Radioberry-2.x #57: why not share RPI4 full image that "just works"?

bieganski, 2024-04-21:
hello. i would like to share my thoughts and suggestions.

the Radioberry project itself is cool, however i found *many* issues when trying to build it. namely:

* `pihpsdr_install.sh` and friends always download newest version of repositories.
* seemingly newest raspberry pi OS (with newest kernel) is assumed. it's a pain to find a working one, and the fact that it is shipped in few configurations (32bit, 64bit) does not help.

i guess that most people (like i do) don't care what version of Linux runs under the hood. people want stuff that "just works"

pa3gsb, 2024-04-21:
The radioberry is not a project you should start when you want a finished product.

I  started the project for myself. Decided to publish everything i have learned. 
Iam not selling anything, not making any money with it. it is all open soft and hardware.

So if you feel the urge to have an image please do!

Using a non linux solution... look to the radioberry juice.

Iam looking forward to try the image.

If you

f4exb, 2024-05-27:
Instructions are missing the libfftw3-dev dependency. 

Also it fails when compiling pihpsdr because it does not find wdsp.h. Just cd to wdsp directory and do "sudo make install" thus the include gets installed. cd back to the root folder and do "make -j$(nproc)" again then it will compile successfully.

schnuvany, 2025-05-06:
Hi Johann, 
https://github.com/pa3gsb/Radioberry-2.x/releases

You write:
[Radioberry Software Installation @Juice](https://github.com/pa3gsb/Radioberry-2.x/releases/tag/radioberry-juice) Pre-release
Installation of radioberry juice software.
This release is only applicable for radioberry with Cyclone CL025.

I´m sorry i have CL016.
I´m also sorry: My set Display+Raspy4b+Radioberry is no free place for extra Board.

schnuvany, 2025-05-06:
Hi all,
i do on a fresh img SC-card on Raspy4b:
https://github.com/pa3gsb/Radioberry-2.x/releases/tag/radioberry

cd /tmp
wget https://raw.githubusercontent.com/pa3gsb/Radioberry-2.x/master/SBC/rpi-4/releases/dev/radioberry_install.sh
sudo chmod +x radioberry_install.sh
./radioberry_install.sh

i solved all missing files, i solved apt update , i solved apt upgrade

But i never solved 
sudo radioberry

here i always missing:

Radioberry gateware version .... 

Thanks

pa3gsb, 2025-08-15:
no one stood up so no need for an image i suppose.

closed 2025-08-15 by pa3gsb