pa3gsb/Radioberry-2.x #60: Error with kernel headers

Montilien, 2024-06-12:
Hi,
I have tried to launch install script on several RPI4 distro, and everytime, install fails with this message:

Radioberry gateware Cyclone 10 CL016 installed.
Installing Radioberry driver...
modprobe: FATAL: Module radioberry not found.
make[1]: *** /lib/modules/6.6.20+rpt-rpi-v8/build: No such file or directory.  Stop.
make: *** [Makefile:8: radioberry.ko] Error 2
Radioberry driver installation failed.

You are using a linux version without linux header files; choose an other distro.
I have tried distros Bullseye and Workworm, 32 and 64 bits and result is the same.
If a specific realease of raspberry pi OS is needed please, give me the information...
Thanks for your help
73's Andre
F1FHK

pa3gsb, 2024-06-24:
Andre

I still not aware why this is still a problem.  Can you give me what you have done when installing.

Info from raspberry pi : https://www.raspberrypi.com/documentation/computers/os.html

I did make video showing bullseye but you should use bookworm.

https://groups.google.com/g/radioberry/c/BILlnTU8YUk/m/ryRlNYTPAQAJ

Let  me know if this helps.

Johan

keyboarderror, 2024-09-02:
I just updated my kernel using rpi-update and I now have the header error. Is there an easy way to fix this?

closed 2025-01-30 by pa3gsb