pa3gsb/Radioberry-2.x #76: Install failure

G0MJW, 2025-07-26:
Latest PI OS 64 bit downloaded 30 minutes ago

Running the script for CL025

The script fails with a message that the OS does not include the linux headers... use a different distribution

So what distribution should we be using?

doalex, 2025-08-08:
Worked for me on a fresh PI5, after applying the fixes from https://github.com/pa3gsb/Radioberry-2.x/issues/75

G0MJW, 2025-08-09:
It might have worked for you - but not very helpful as it didn't work for me. If you really wanted to be helpful; you might explain what OS and Kernel and be a little less brief with the descriptions. We are not all professional developers. Anyway, it works fine with the PI 4 so problem solved. The PI 5 is not an upgrade here.

closed 2025-08-09 by G0MJW